RigoLigoRLC/LC2KiCad #11: Missing closing parenthesis and LF char

Cricko25, 2020-11-05:
[key.json.gz](https://github.com/RigoLigoRLC/LC2KiCad/files/5496560/key.json.gz)

Conversion of the key.json produces the following _incomplete_ line

(fp_line (start 115.901724 31.582868) (end 115.901724 26.582878) (width 0.050038) (layer F.Cu)  (fp_line (start 115.901724 26.582878) (end 106.601768 26.582878) (width 0.050038) (layer F.Cu) (fp_line (start 106.601768 26.582878) (end 106.601768 31.582868) (width 0.050038) (layer F.Cu)  (fp_line (start 106.601768 31.582868) (end 115.901724 31.582868) (width 0.050038) (layer F.Cu))
  
without LF -- that is, "(fp_line..." is from the next line.

Manually inserting ")<LF>" before " (fp_line" makes the file correct.  Perhaps it is forgotten to be issued from the program.

Similar situation for a different object type:

[res.json.gz](https://github.com/RigoLigoRLC/LC2KiCad/files/5496631/res.json.gz)
  (segment (start 17.080078 -19.558000) (end 17.080078 -21.652078) (width 0.254000) (layer F.Cu)  (segment (start 17.080078 -21.652078) (end 17.526000 -22.098000) (width 0.254000) (layer F.Cu)  (segment (start 17.526000 -22.098000) (end 18.669000 -22.098000) (width 0.254000) (layer F.Cu)  (segment (start 18.669000 -22.098000) (end 18.480126 -21.909126) (width 0.254000) (layer F.Cu)  (segment (start 18.480126 -21.909126) (end 18.480126 -19.558000) (width 0.254000) (layer F.Cu)(net 1))

Adding "(net 1))<LF>" before "(segment (start ..." solves this.

RigoLigoRLC, 2020-11-06:
[image]
This is so freakin stupid.

closed 2020-11-06 by RigoLigoRLC

Cricko25, 2020-11-06:
*LOL* are you used to write in Python? :-) never mind, grats you killed this bug and thanks for the program

RigoLigoRLC, 2020-11-06:
No, this originally comes from a bug fix issued on my phone. And the plain
old Vim didn't really get me to know about this issue. :p

commit b60f1b40 (2020-11-06): Fix #11: add the missing brackets

This is a fucking stupid issue, duh.
commit 19c5d190 (2020-11-12): Fix #11 caused by fixing #9: add missing brackets

This is a fucking stupid issue, duh.