RigoLigoRLC/LC2KiCad #18: window 下编译：cmake报错

multigram, 2021-07-31:
你好，按照说明用mingw&cmake编译，失败信息如下，请帮忙解答一下，谢谢！

WeiFan@WeiFanPC MINGW32 /c/Users/WeiFan/Downloads/LC2KiCad-master/LC2KiCad-master/build
$ cmake -G "MinGW Makefiles" .. && mingw32-make -j 8
CMake Error: CMake was unable to find a build program corresponding to "MinGW Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMake was unable to find a build program corresponding to "MinGW Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!

RigoLigoRLC, 2021-07-31:
执行cmake
--help，在最后会输出CMake支持的generator。根据实际情况，选用合适的generator，比如某种Makefile或Ninja（如果有安装）。

multigram, 2021-07-31:
谢谢支持

closed 2021-07-31 by multigram