RigoLigoRLC/LC2KiCad #20: Conversion fails with 1675 errors

reinauer, 2022-01-21:
Tried this on one of my more complex, working boards.

```
[Auto Parser] Read input document "/home/stepan/Desktop/1-PCB_PCB_amiga.json" as EasyEDA 6 document...
Warning: gge172276: Flood fill keepout regions will prevent all fills rather than just flood fills with lower priority. This is a behavior difference. You've been warned.
Warning: gge168972: Flood fill keepout regions will prevent all fills rather than just flood fills with lower priority. This is a behavior difference. You've been warned.
Warning: gge168807: Flood fill keepout regions will prevent all fills rather than just flood fills with lower priority. This is a behavior difference. You've been warned.
Info: gge129259: Flood fill area spoke width was not set; it is now 0.508mm.
Info: gge129266: Flood fill area spoke width was not set; it is now 0.508mm.
Info: gge157801: Flood fill area spoke width was not set; it is now 0.508mm.
Info: gge157882: Flood fill area spoke width was not set; it is now 0.508mm.
Info: gge75974: Flood fill area spoke width was not set; it is now 0.508mm.
Warning: gge175624: Flood fill keepout regions will prevent all fills rather than just flood fills with lower priority. This is a behavior difference. You've been warned.
Info: gge175627: Flood fill area spoke width was not set; it is now 0.508mm.
Warning: gge75981: Flood fill keepout regions will prevent all fills rather than just flood fills with lower priority. This is a behavior difference. You've been warned.
Info: gge175711: Flood fill area spoke width was not set; it is now 0.508mm.
Info: gge75977: Flood fill area spoke width was not set; it is now 0.508mm.
Warning: gge158393: Flood fill keepout regions will prevent all fills rather than just flood fills with lower priority. This is a behavior difference. You've been warned.
Info: gge175492: Flood fill area spoke width was not set; it is now 0.508mm.
Info: gge175495: Flood fill area spoke width was not set; it is now 0.508mm.
Warning: gge168643: Flood fill keepout regions will prevent all fills rather than just flood fills with lower priority. This is a behavior difference. You've been warned.
Warning: gge168642: Flood fill keepout regions will prevent all fills rather than just flood fills with lower priority. This is a behavior difference. You've been warned.
Warning: gge148347: Flood fill keepout regions will prevent all fills rather than just flood fills with lower priority. This is a behavior difference. You've been warned.
Warning: gge75968: Flood fill keepout regions will prevent all fills rather than just flood fills with lower priority. This is a behavior difference. You've been warned.
Warning: gge148342: Flood fill keepout regions will prevent all fills rather than just flood fills with lower priority. This is a behavior difference. You've been warned.
Warning: gge148344: Flood fill keepout regions will prevent all fills rather than just flood fills with lower priority. This is a behavior difference. You've been warned.
Warning: gge148346: Flood fill keepout regions will prevent all fills rather than just flood fills with lower priority. This is a behavior difference. You've been warned.
Info: gge157465: Flood fill area spoke width was not set; it is now 0.508mm.
Info: gge157791: Flood fill area spoke width was not set; it is now 0.508mm.
Info: gge157923: Flood fill area spoke width was not set; it is now 0.508mm.
Info: gge166926: Flood fill area spoke width was not set; it is now 0.508mm.
Error: gge175766: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge175769: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge175772: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge175775: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge175778: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: An SVGNODE object has been discarded.
Error: gge175868: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge175871: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge175874: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge175877: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge175880: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: An SVGNODE object has been discarded.
Error: gge174769: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: An SVGNODE object has been discarded.
Error: An SVGNODE object has been discarded.
Error: gge172308: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge172311: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge172314: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: An SVGNODE object has been discarded.
Error: An SVGNODE object has been discarded.
Error: gge171040: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge168209: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge168210: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge168211: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge168212: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge168213: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge168214: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge168223: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge168224: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge168225: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge168226: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge168227: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge168228: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge166124: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge166125: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge166126: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge166127: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge166128: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: An SVGNODE object has been discarded.
[... 1625 lines cut ...]
Error: gge21591: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge21592: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge21593: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge21594: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: An SVGNODE object has been discarded.
Error: gge21556: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge21557: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
Error: gge21558: A fill region was found inside a footprint, which is not allowed in KiCad. This region is discarded!
[... cut ...]
Warning: gge160811: Copper track on footprint. This is not recommended.
Warning: gge160823: Copper track on footprint. This is not recommended.
Warning: gge160835: Copper track on footprint. This is not recommended.
Warning: gge160847: Copper track on footprint. This is not recommended.
Warning: gge160859: Copper track on footprint. This is not recommended.
Warning: gge160871: Copper track on footprint. This is not recommended.

Warning: Error(s): 1675, warning(s): 21.
```

Trying to load the resulting file, kicad dies with "Un-terminated delimited string"

RigoLigoRLC, 2022-01-21:
Ah, most of the errors are about discarded unsupported objects. But the bug should be unrelated to these objects. If you can provide the design file then I can tackle it down.

RigoLigoRLC, 2022-02-02:
Could you please provide an EasyEDA file that can reproduce this issue? As I'm no longer actively maintaining this tool, I don't think I'm going to find out the bug without your help.

reinauer, 2022-02-02:
I will send the files to you separately. Thank you for your help on this.

RigoLigoRLC, 2022-03-20:
Closing as the failure of conversion is solved.

closed 2022-03-20 by RigoLigoRLC